Lramseyer/vaporview #14: Different signals are displayed in different styles

narutozxp, 2024-07-06:
Dear Lramseyer
This plug-in is easy to use, but the currently set style takes effect globally. It would be nice if each signal could have its own independent display style. Because many times, one signal needs to be displayed in hexadecimal, but other signals need to be displayed in decimal.

Lramseyer, 2024-07-10:
Great idea! Something I have actually been planning on implementing, but haven't gotten around to yet. I was thinking of defaulting to hexadecimal, and using a context menu (right click menu) to set anything else.
Thoughts?

narutozxp, 2024-07-11:
I think this is a good way

Lramseyer, 2024-07-17:
OK, this should be fixed in the [1.0.1](https://github.com/Lramseyer/vaporview/releases/tag/release-candidate) release. Check it out!

Lramseyer, 2024-07-17:
Sorry accidentally hit close!

Lramseyer, 2024-07-26:
1.0.1 is released to the extension marketplace with this fix in place!

closed 2024-07-26 by Lramseyer