Lramseyer/vaporview #18: Support protocol decoding

hongshui3000, 2024-08-03:
It would be great if the waveform viewer could support standard protocol decoding (such as i2c, uart, etc.), or if a waveform decoding plug-in could be added.
For example, it can support the following decoding library. It will be more convenient for analyzing some standard protocols
https://github.com/sigrokproject/libsigrokdecode

Lramseyer, 2024-08-07:
Hmmm... This isn't a bad idea. One thing I want to create is an "annotations API" and I'll probably create a larger discussion thread for the topic. It's essentially this and more. What I ultimately want is for people to be able to create custom overlays for signal values and groups of signals. The idea would be that other extension developers could create these or incorporate them into vaporview.

As an example, the first one I am working on is being able to parse .objdump files and link a program counter to an .objdump location to have better visibility into which line of code is being executed.

What I need to work out though, is how I should organize that information and communicate between APIs. I plan on migrating file parsing to the wellen library, so the waveform data structures are subject to change.

ArcSpecter, 2025-02-21:
> It would be great if the waveform viewer could support standard protocol decoding (such as i2c, uart, etc.), or if a waveform decoding plug-in could be added. For example, it can support the [...]

Sorry but this is an excessive request for a typical tool intended to view logic signals in digital sim. Such kind of feature is common for software hooked to Logic Analyzers, but this is not such a tool and serves different purpose. Thus it would be wise to spend resources on addressing other issues which are more relevant for logic sim signal view software.