Lramseyer/vaporview #19: Support Linear, and Stepped render modes

hongshui3000, 2024-08-03:
Hope to support the same mode as WaveTrace extension.
These modes are useful for viewing waveforms.
[image]

Lramseyer, 2024-08-03:
I'm going to be honest, I don't actually have any VCD files with analog data in them to test this out ...which is a little ironic considering all of the mixed signal work I do in my day job!

I'll look around at how analog data is stored in a VCD file, and I should be able to add this to VaporView pretty easily!

hongshui3000, 2024-08-05:
It does not necessarily require analog data. For example, in the picture above, it is just the value of a counter, which can form a linear/Stepped  waveform.This kind of graph is very useful for finding differences in a large amount of similar data.

Lramseyer, 2024-08-06:
OK, so it sounds like you're asking for multi bit waveforms to be viewed as analog data? I should also support 2's compliment while I'm at it.

tafkos, 2024-11-03:
https://edaplayground.com/x/sf2W - a simple verilog code to produce VCD with sinewaves. Both real and regs are generated.

Lramseyer, 2025-02-07:
Thanks @tafkos for the test file! It really helped. These features are in the 1.3.0 pre-release build. Give it a try and let me know what you think.

I'll get to the 2's compliment later. It's a pretty trivial update.

Lramseyer, 2025-02-12:
Fixed in 1.3.0

closed 2025-02-12 by Lramseyer