Lramseyer/vaporview #20: Add rtlbrowse stems file support

Lramseyer, 2024-09-09:
# stems file support

I would like to add an [rtlbrowse ](https://linux.die.net/man/1/rtlbrowse) like feature into VaporView that uses stems files to link netlist items back to RTL. This looks pretty straightforward all things considered.

The only problem is that I have never used verilator or rtlbrowse, since I have always used VCS/Verdi. The only example I have is [here](https://github.com/gtkwave/gtkwave/blob/master/examples/des.stems) If anyone wants to donate to the cause, or has an example repo that they can point me to, that would be a huge help!

## How I plan to implement it

I will add a right click [context menu] for stems files to import them into an active viewer. Once it loads, it will parse the file, and create an internal netlist and check it against what's loaded into the viewer. It will discern the RTL path and parent module, and line number in the file. Once that's loaded in, I can have right click menus to view the item in the RTL. Though this may be tricky, since it looks like the stems files only point to where the module is defined.

## Other things to look into

- Index all of the file locations, so that from the RTL, you can place signals into the viewer, but that might have to be a stretch goal for now.
- Driver/Load Tracing - I might have to rely on another RTL syntax extension for this. If any of you are owners or contributors to any of these extensions. Please reach out!

ecstrema, 2025-02-18:
Wow that would be truly incredible.

Lramseyer, 2025-08-22:
I'm going to close this issue, since this will be implemented differently. Take a look at this project, as it interacts with the [Vaporview API](https://github.com/Lramseyer/vaporview/blob/main/API_DOCS.md).

[https://github.com/heyfey/sv-pathfinder](https://github.com/heyfey/sv-pathfinder)