Lramseyer/vaporview #22: Display individual bits from a wide signal

krisjdev, 2024-10-15:
Currently VaporView only allows you to display signals with their full bit width, but in some scenarios it is helpful to split them up and display the individual bits. For example, eight 1-bit signals may be packed into one variable (``ui_in[7:0]``), but you can't display ``ui_in[4]`` by itself.

It would be nice if it had functionality similar to GTKWave, where you can double click on a packed variable to expand it, then copy-paste the signals elsewhere with the option to rename them for clarity. Or possibly using the netlist viewer to further expand the signal down into individual bits, and checking/unchecking the ones that you want displayed.

[image]

<br>

Also just wanted to say that this project is very cool :) It integrates very nicely with VSC, and I can't wait to replace GTKW with it.

Lramseyer, 2024-10-15:
This is definitely something I have thought about and plan to implement. I will probably allow users to expand multi-bit signals as if they were modules. My only concern is that I want to make sure that the user knows that multi-bit signals are selectable (because modules are not.)

What I might do, is require the user to right click, and select a menu option to expand that signal. But that might be too much friction if that's something that a user wants to do often. The only time I really ever have to do this is wen I want to break out an interrupt or error flag bus. But my work lately has been on small MCUs.

Lramseyer, 2024-11-09:
I made a whole bunch of changes to the netlist view, and I have laid the groundwork for this feature. I removed the checkboxes for scope items, and if the waveform dump breaks out the individual bits as separate variables, Vaporview now groups them together. What you're specifically asking for won't make it into the 1.2 release in its entirety, but progress is being made! I promise!

krisjdev, 2024-11-09:
That's awesome to hear!! I can see the auto-grouping being very useful - [LogiSim Evolution](https://github.com/logisim-evolution/logisim-evolution) for whatever reason dumps each multi-bit signal as individual bits if you're targeting an FPGA.

I do agree that having to right-click to open the signal might be a bit tedious, although if there's a saved state then I wouldn't mind doing the initial setup. As long as you're not making the user do it every time VSC loads then it should be okay, I think.

Lramseyer, 2024-11-15:
Hey, so bad news: I have to de-feature this because of a [VScode bug](https://github.com/Lramseyer/vaporview/issues/27).

In essence, I use TreeDataProviders for my TreeViews with the manageCheckboxStateManually = true. However, there's a bug with the latest version of VScode where onDIdChangeCheckboxState events don't populate when manageCheckboxStateManually is set to true. If I set it to false, I have to remove all checkboxes for TreeItems that have child Items (otherwise it will check all of the child items, and we don't want that.) This is fine for scope items, but it breaks your feature.

Furthermore, I didn't remove checkboxes in 1.1.0, and manageCheckboxStateManually is set to true, so this bug breaks the old version. In the short term, I would rather fix things for existing users rather than wait on the VScode team to get a release out. Apologies for the inconvenience. Once this bug is fixed, I will add this feature back into a 1.2.1 release.

If you would like, I can create a custom build with this feature enabled for you, or walk you through how to build it yourself.

krisjdev, 2024-11-18:
Ah, that's quite unfortunate. To be honest I'm just a hobbyist, so I don't think it would be wise for me to use your time for something that I would only use for a couple sessions. I'm more than happy to just wait until it gets properly released :)

tafkos, 2025-03-06:
> Once this bug is fixed, I will add this feature back into a 1.2.1 release.

@Lramseyer #27 seem resolved... Can you reenable this powerful feature?

ilnerdchuck, 2025-03-24:
Is this feature still disabled in 1.3.1?

[image]

It will be really useful to have, I'm really struggling with signal arrays xD

Lramseyer, 2025-04-18:
The [1.3.3 Release](https://marketplace.visualstudio.com/items?itemName=lramseyer.vaporview#review-details) is out now!

Lucretia, 2025-11-30:
I'm trying to follow VHDL Whiz (currently on video 11) with GHDL and I have tried dumping to VCD and GHW and I cannot seem to separate out the array into individual bits, what am I missing?

Lramseyer, 2026-02-18:
I know things have been a bit quiet on this issue here for a while, but I haven't forgotten about this, and I finally implemented first class support for this in the viewer! It's available on the 1.5.0 dev builds, which you can download [here](https://github.com/Lramseyer/vaporview/releases/tag/1.5.0-dev). Let me know what you think about this feature.

krisjdev, 2026-02-23:
Thank you so much for the all work you've done on this :) I've checked out the dev build you linked and noticed a couple things:

1. Clicking on "Create Signals from Bits" twice (without renaming the first group) freezes the waveform viewer
2. Creating a signal from bits doesn't respect the bit order (i.e. `signal[7:0]` is sorted from 0 up to 7, instead of 7 down to 0)
3. Reversed bit slices fail to render a waveform (see `tb.ui_in[0:3]` in the image below)

[image]

4. Vaporview didn't save settings with the `.json` extension appended, meaning I couldn't select the newly created save file without adding the extension myself via the file browser

Small thing with point 4 - is there a way to stop it from prompting to save immediately?

The points above are just small quality-of-life things which can be sorted with a little bit of effort from me, so other than that, it looks great!

Lramseyer, 2026-02-23:
Thanks for checking it out! It should be a pretty easy fix for Issues 1, 2, and 4. For issue 3, that will take me some time to fix, and I will probably not add that to 1.5.0 since that will take me more time and I want to get this version out to users. Can you create a separate feature request for that?

Also, it looks like you have a renderer glitch. What OS are you using, and what is your zoom level?

Lramseyer, 2026-02-23:
> Yeah, the timescale text is a little blurry - I'm on Linux Mint 22.1 Cinnamon, and the window isn't zoomed in/out (I reset with the View: Reset Zoom command), but in my vscode settings I have window.zoom set to -1. Happens on both 1440p and 1080p monitors if that's of any use to you?

That's strange. Does it go away if you resize the viewer? If not, it sounds like a bug with window.devicePixelRatio. Either way, it will probably be fixed in newer versions, as I plan to update the multi-bit renderer.

Lramseyer, 2026-02-23:
OK, those were pretty easy fixes. I pushed them and updated the builds on the [downloads page](https://github.com/Lramseyer/vaporview/releases/tag/1.5.0-dev). If nothing major comes up, I'll probably release on Wednesday.