Lramseyer/vaporview #30: Extension doesn't display signal on VSCode web

mdanesh, 2024-12-10:
On web (vscode.dev) the extension doesn't render the waveform.  Same waveform renders OK on desktop version of VSCode. 

**To Reproduce**
1. Download the following file https://vc.drom.io/?github=wavedrom/vcd-samples/trunk/swerv1.vcd
2. Load and display in Desktop version -> Works fine 
[image]

4. Open vscode.dev on web browser (Chrome or Safari) 
5. Load the same VCD -> no signal display.  The code editor attempts to open and fails.  
[image]
[image]

Does VaporView actually support VCCode web?   Is there a setting that is missing? 

Thanks in advance

Lramseyer, 2024-12-10:
It doesn't support VScode web anymore, _but_ I should be able to add that back in. I removed that support in version 1.1 when I moved all of the file handling over to node's fs library to support larger file sizes. I could implement a fallback method that supports web, but it would be limited in file size for waveform dumps. That won't be as much of a problem anymore now because it supports FST dumps.

In the meantime, you can try using version 1.0 or 1.0.1. It's far less optimized and polished, but it supports loading more than 8 variables without having to pay money!

If you want to look into creating a custom build that supports web, I would suggest looking in src/extension_core/document.ts, and commenting out the import of fs. Use vscode.workspace.fs instead. Then you would need to set loadStatic to true always. You might need to modify some of the build targets from node to web, but that should work.

I am trying to stabilize some changes and refactoring for the 1.2.5 release, but I'll keep this issue open as a to-do item, as adding web support is something I have been meaning to add back in!

mdanesh, 2024-12-11:
Thanks for the quick feedback.  What is the file size limit (VCD and FST) that you expect the web version would have?  
The availability of this functionality on web is pretty useful. I'll keep a watch for when you can fold that in. Thanks.

Lramseyer, 2024-12-11:
I'm not sure what the limit is for web, but I know that the vscode implementation of fs is limited to 2GB.

In my experience, FST files are significantly smaller than VCD files (VCD is uncompressed plaintext, FST is binary, and compressed) and I have seen a 50x difference between the two formats. If you're using a FST file larger than 2GB, than you probably work for a semiconductor company.

Lramseyer, 2025-01-29:
OK, I added a test build in the [releases](https://github.com/Lramseyer/vaporview/releases/tag/1.3.0-pre-release) section. Let me know if that fixes your issue!

pengchengduan, 2025-02-11:
Thanks a lot! I'm looking for the same requirement

Lramseyer, 2025-02-12:
Can someone go check with the 1.3.0 release? I believe this was fixed, but I need someone to confirm! 

@mdanesh or @pengchengduan

pengchengduan, 2025-02-12:
It seems that it still does not work. 
I install the 1.3.0 vsix, below is when I open the vcd file.

[image]

[image]

pengchengduan, 2025-02-12:
I use the code-server

[image]

Lramseyer, 2025-02-12:
It looks like it's loading the netlist, so it's at least loading the file.

How big of a VCD file is this?
What happens when you try to add variables into the viewer (you have to check the box)?

pengchengduan, 2025-02-12:
the file is about 13mb,  I also load a smaller file(1.2k) 
I can not add variables. vscode will ignore my checking the box

[image]

pengchengduan, 2025-02-12:
I use a smaller vcd file about 1.2k
The code-server ignores my checking the box

[image]

Lramseyer, 2025-02-12:
OK, it looks like it's a different issue than I thought. It's properly parsing the file, but it has trouble actually creating the webview. I know this because you're able to expand the netlist tree (which means it finished parsing the value change data), and it's un-checking variables for you.

I initially thought this was an [fs](https://nodejs.org/api/fs.html) issue, since the web version doesn't support node fs (which I need so that I can read files larger than 2GB) But it looks like it's parsing the file. Let me go play around with VScode web. I have no debug flow for it, so please bear with me as I figure out how to do that!

pengchengduan, 2025-02-12:
I tried several plugins(wavetrace, digital-ide,surfer), unluck they all do not work on the web version.
It will be great if vaporview supports. Thanks!

Lramseyer, 2025-03-14:
I haven't forgotten about this! It looks like it's not liking the worker_threads package. This is necessary because it's what allows the WebAssembly component to work. I'll have to see if there's some web compatible version of this package that I can use in its place, but if anyone in the community has any ideas, I'm all ears!