Lramseyer/vaporview #33: Sidebar issue

Ansh-Srivastav2023, 2025-01-09:
- OS: [e.g. Windows]
 - Browser [MS Edge]

Whenever, I try to move a signal, click anywhere on the waveform or on any signal, zoom-in or out, or change the data format (binary, decimal, etc.) the sidebar keeps on appearing again and again. Once I remove the sidebar, and as soon as I perform some operation on the waveform (as mentioned above), the sidebar reappears (Which is kind of irritating too😁). What I expect is the sidebar to appear only if I give some command, intentionally, for it to appear. Please...🙏

[image]

https://github.com/user-attachments/assets/bbfe1b43-5109-4e68-b07d-83525e03c5fd

Lramseyer, 2025-01-09:
OK, so first of all, nice editing!

Regarding to your problem, this may be a VS Code issue that is out of my control as an extension developer. But I suspect that this may be due to the extension calling TreeView.reveal() every time you select a signal. I believe I implemented that in 1.2.5. Could you try using version 1.1 or 1.2 and seeing if the problem persists? If it goes away, I will remove TreeView.reveal() (as it isn't really necessary) I'm getting pretty close to the 1.3 release, and before I do that, I will be uploading test builds in the releases section before it goes live.

Ansh-Srivastav2023, 2025-01-09:
I really appreciate you getting to me so quickly. Thanks a lot.

The problem persists with v1.2.0 and 1.2.5, but works fine with v1.1.0.

Lramseyer, 2025-01-09:
OK good news - I was able to recreate this issue on the native version of VScode, and it looks like TreeView.reveal() is what's causing this. Essentially, any time you select a variable in the viewer (there are other events that can also trigger this,) the Displayed Signals view highlights the selected variable, and because of this, the side bar keeps reappearing. So in version 1.3.0, I have added a "Show in Netlist View" context menu option, which is a more sensible place to have the TreeView.reveal() function, as it highlights the variable in the netlist view.

...heck, I might remove the "Displayed Signals" View entirely, since there are so many other easier ways to remove a variable from the viewer that I have added since 1.0.

I'll create test builds soon. Hopefully this helps you out in the interim!

Lramseyer, 2025-01-10:
OK, I added a test build for 1.3.0. It's missing a few features, which I am still in the process of adding, but it should fix your issue!

Lramseyer, 2025-02-12:
Fixed in 1.3.0

Lramseyer, 2025-02-12:
Fixed in 1.3.0

closed 2025-02-12 by Lramseyer