Lramseyer/vaporview #35: Apply "Add all signals in module" recursively

justarandomgeek, 2025-01-14:
I'm not sure if this is most properly classified as a bug report or a feature request, but i'm working with a trace with a hierarchy that has many deep but narrow branches (captured Factorio circuit signals), and it would be nice if "Add all signals in module" would apply recursively from the higher levels so i don't have to open them all the way down every time!

[image]

Lramseyer, 2025-01-14:
Good idea! I should be able to put that in pretty easily.

Lramseyer, 2025-01-15:
OK, I just added this feature. It's in the [pre-release build](https://github.com/Lramseyer/vaporview/releases/tag/1.3.0-pre-release) if you want to give it a try!

Lramseyer, 2025-02-12:
Fixed in 1.3.0

closed 2025-02-12 by Lramseyer