Lramseyer/vaporview #36: "Add all signals in Module" does not work unless module is expanded in tree

fake-name, 2025-01-29:
**Describe the bug**
The right-click -> "Add all signals in Module" option does not work unless the module entry in the netlist tree is also expanded.

**To Reproduce**
1. In the netlist panel
2. _Without left clicking first_, right click on an netlist subgroup
3. Click "Add all signals in Module"
4. Nothing happens

[image]

Note that this seems to be weirdly flaky. I've had it work a few times, and I've got no idea why.

There is a more broad version of this that is also broken, namely that if you select multiple entries in the netlist view, and try to add them at the same time, nothing happens there either.

[image]

**Expected behavior**
I would expect hitting "add all signals to module" to add all the signals contained in the object to the view.

 

**Desktop (please complete the following information):**
 - OS: W10
 - VSCode:
        
        Version: 1.95.1 (system setup)
        Commit: 65edc4939843c90c34d61f4ce11704f09d3e5cb6
        Date: 2024-10-31T05:14:54.222Z (3 mos ago)
        Electron: 32.2.1
        ElectronBuildId: 10427718
        Chromium: 128.0.6613.186
        Node.js: 20.18.0
        V8: 12.8.374.38-electron.0
        OS: Windows_NT x64 10.0.19045
 - Plugin version: Current VSCode plugin (1.2.6, I think?)
 

**Additional context**
There's the more broad issue that I cannot select the root node in the signal tree, and add _all_ the signals in the entire file to the visualizer in one step, but I suppose that's being solved in https://github.com/Lramseyer/vaporview/issues/35. Really, right now "Add all signals in module" is incorrect, it's more of "add all signals in immediate module container".

Lramseyer, 2025-01-29:
OK, I just checked, and I already fixed it in my dev builds. Feel free to download the [1.3.0 Pre-release](https://github.com/Lramseyer/vaporview/releases/tag/1.3.0-pre-release) and install it, as that should have the fix in it. If you're feeling really brave, feel free to clone and build the extension yourself.

I also changed the nomenclature to variable/scope from signal/module.

Lramseyer, 2025-02-12:
Fixed in 1.3.0

closed 2025-02-12 by Lramseyer