Lramseyer/vaporview #38: Look into porting rendering pipeline to canvas

Lramseyer, 2025-02-12:
Right now, the rendering pipeline creates chunks that are dynamically loaded in and out as the user scrolls on the page. Each chunk is an HTML element with separate HTML elements for each signal in the viewport, and the waveforms are SVGs drawn in chunks. The idea behind this was to minimize processing power required for the renderer. I also tried to make it render chunks independent of when a frame would draw so that performance could be smooth even if a chunk didn't render in time. But that wasn't how it actually ended up working in practice. I made this decision before I knew what I know now about HTML5 rendering.

Based on my profiling (on my old windows laptop) it takes about 2-5ms to render a chunk with a dozen or so waveforms and insert them into the DOM. Presumably, I should have some time savings by not having to deal with DOM manipulation AND not converting my paths to strings. On the flip side, I will be drawing a bigger "chunk", which might take more time. Since I will be doing this every frame that the user scrolls or resizes, I will need to keep render times very low.

Work for this will be done on a canvas-render branch. If anyone has thoughts on the matter, please speak up, as I do not have much experience with canvas!

ecstrema, 2025-02-18:
First of all, great job on vaporview, it's really a pleasure to work with. Keep in mind that I have no idea of how the internals of vaporview work in this next comment, so some things might already be implemented, and others incompatible with the current rendering model.

Canvas drawing is considerably more tedious than svg though. It would definitely be faster, but from what I saw in [surfer](https://surfer-project.org), it makes interactivity with the signals very verbose and error prone.

Before taking speed into considerations though, the usual problem with large waveforms is the amount of memory needed. Fortunately, as long as it stays in rust land, this is handled very nicely by wellen.

So a few questions before knowing if moving to canvas is worth it: is the svg generation happening on rust side? Are the signals value sent to js at all? Ideally, for both performance and memory, js should just receive an svg string, put it into the DOM, and make it interactive (hydrate it).

SVG can be better than canvas when properly used as it allows the renderer to automatically use the GPU for some tasks, such as animations.
 - Minimize the changes in the SVG DOM element itself
   - Pan and zoom with the viewbox attribute as much as possible
   - draw text items above, because they are not static
   - Draw the cursor above
 - Use a single path element per differentiable item (do you want to be able to select with your mouse a single value of a signal, or is selecting the whole signal enough?)

For long simulations, rendering the whole timeline to svg might be too much, so the same technique used by maps renderers might be a possible option: tiles: The renderer should be able to render tiles of a specific timeframe, leaving the rest empty. Tiles can then be dynamically added to the SVG.

Once again, thanks for vaporview.

Lramseyer, 2025-02-20:
Believe it or not, but the rendering pipeline is currently implemented in Javascript with SVGs, and I tile things as you suggested. It works pretty well, but it isn't good for dragging the scrollbar, as it takes a few frames to load in and out due to the way I do asynchronous rendering. If it tries to draw too many chunks, it gets laggy otherwise.

What I am planning to do with canvas is to use a separate canvas for each waveform. This way, I can still capture events and maintain VScode context for each displayed signal. I have a branch that I am working on. it's coming together pretty nicely, and once I get all of the old chunk rendering code removed and cleaned out, I'll push it so that you can take a look at it. The changes will be in src/webview/renderer.ts and src/webview/viewport.ts.

Another added bonus to all of this is that the value display text in the multi bit waveform values actually centers and aligns properly without a bunch of cursed math to account for chunk boundaries. I really hope canvas works out if only for this reason!

ecstrema, 2025-02-21:
Sounds good!

Having different canvas for each signal definetely sounds like a win, I hope the text drawing won't be too tedious!

I am very curious about the memory usage of this method for large files. 

Good work! Thanks again for the extension!

Lramseyer, 2025-02-22:
> I am very curious about the memory usage of this method for large files.

I am not sure what the canvas memory usage looks like, but this is running on electron at the end of the day. Personally, I would have called it Neutron or top quark because it's not very light weight.

As far as value change data goes, only the signals loaded into the viewer are stored in memory (for FST files at least) For VCD files, the data is compressed in memory. But users should really know better than to use large VCD files.

ecstrema, 2025-02-24:
After some more thinking and [experimentation](https://ecstrema.github.io/swell/), canvas seems like a good choice. Since you'll only be iterating over the value changes from wellen, this can be done without any copying and so should be able to handle large amounts of data. Way better than svg actually, which needs to copy the whole data into the svg, instead of simply going through it once for rendering.

> Personally, I would have called it Neutron or top quark because it's not very light weight.

Haha, as someone who currently works at CERN for beam instrumentation, this is very funny. However, although the tree views for selecting the signals depend on vscode, most of the actual rendering is not, right? So this may be able to be extracted to Tauri or equivalent to get an independant executable?

Lramseyer, 2025-02-26:
OK, this item is more or less complete! I was able to rip out a bunch of code! You can download test builds [here](https://github.com/Lramseyer/vaporview/releases/tag/1.3.1-pre-release).

Lramseyer, 2025-03-02:
1.3.1 is released to the marketplace, so I am marking this as complete!

closed 2025-03-02 by Lramseyer