Lramseyer/vaporview #39: Auto import signal list

ecstrema, 2025-02-18:
When opening a waveform, it would be great if vaporview could automatically open adjacent signal lists in the filesystem. 

Basically, I keep opening the waveform file and then loading the associated signal list. These two steps are always the same, and I wish the second was automatic.

Lramseyer, 2025-02-21:
Are you suggesting loading the viewer settings file?

Because that is something on my to-do list. I would like to have something that VScode recognizes so that I can load the viewer settings natively. I want to work with the surfer team to establish a common format so that regardless of what tool you or your coworkers are using, it should work seamlessly.

ecstrema, 2025-02-21:
I am referring to this:

[image]

I would love it if they could be auto loaded when a file is reopened. The same way as the way the cursor position is restored when reopening a file in vscode.

ecstrema, 2025-02-21:
Note however that contrarily to vscode, this should be saved alongside the waveform file, so others can benefit of the setup.

ecstrema, 2025-02-21:
> I want to work with the surfer team to establish a common format so that regardless of what tool you or your coworkers are using, it should work seamlessly.

I would also love that. Surfer currently has a fast evolving format (it's a .ron file), gtkwave also has its format, a lot of others use tcl...

It would be great to have a single basic format on which waveform viewers can expand.

I would propose a format extensible by default with x-[viewer-name], akin to css vendor-specific features. These can be ignored by other vendors, and migrated to new names when these settings stabilize. (x for extension)

```
{
	path_to_waveform: "relative/path", // We currently have absolute paths in both vaporview and surfer
	time_granularity: "ps", // or ns, fs, etc.
	// etc.
	selectedSignals: [
		{
			id: "xpath/to/dut/signal",
			name: "$full" // $default, $full, $short, or custom name
			type: "..."
			"x-vaporview-customSetting": "..."
        },
        // ...
	],
	"x-vaporview-customOtherSetting": "...",
}
```

Lramseyer, 2025-12-01:
OK, so I sort of forgot about this issue, and was a bit unsure on what was being asked, but now this makes sense. Let's see if I can add this to 1.4.4. If it's as easy as I'm anticipating, then I will add it.

Lramseyer, 2025-12-09:
This was pretty easy, I have a test build on the [releases](https://github.com/Lramseyer/vaporview/releases/tag/1.4.4-beta) page.

Now it prompts the user to reload a file if it finds a file in the user's workspace. If you don't like the implementation, I can pretty easily change it.

Lramseyer, 2025-12-12:
Marketplace release is out