Lramseyer/vaporview #40: Name translation file for signal

ArcSpecter, 2025-02-21:
in GTKWave for example, If I want to plot some FSM's signal values, I can assign some textual file with translation of unsigned numbers to the state name, and in the waveform I will have a displayed text which is name for each state.

It is extremely useful and a must have feature, otherwise it is a pain to constantly look up into code for the number of the specific FSM state during debugging.

So is there some way to get this feature in Vaporview? I tried to look how to do it and to me it seems that this can't be done.

If this feature is absent it's definitely should be added because it's what it makes easy to debug digital designs with multiple or large FSMs.

Lramseyer, 2025-02-21:
This is a planned feature, but I didn't have a release target. However, since you asked, I'll be sure to slate this for 1.4!

If you're interested in working on this feature, I would be happy to help you get set up!

ArcSpecter, 2025-02-21:
thanks, unfortunately I'm not a TypeScript person, so I don't think I can be of an immediate help on implementing this feature. If that feature would be there, me (as daytime FPGA dev) would turn Vaporview into a daily driver though.

Lramseyer, 2025-02-26:
Hey @ArcSpecter - do you by any chance have any test files you can send me with enums? Or a link to a github repo works too!

ArcSpecter, 2025-03-01:
Sorry for delayed response, would attaching .vcd file with translation text file suffice?
I am attaching compressed .vcd file as a result of some simulation run, where theres `s_current` signal with 7 states, and the text file with enums of that state.

Please let me know if you need it in some other way, or more details, or how else can I assist you.

[test.vcd.tar.gz](https://github.com/user-attachments/files/19037337/test.vcd.tar.gz)

[xbar_state_t.txt](https://github.com/user-attachments/files/19037340/xbar_state_t.txt)

Lramseyer, 2025-03-01:
OK, so it looks like the enums are in a separate file, which isn't the most helpful. I'll have to take a look, but I think FST supports enums. Do you by any chance have a FST version with the enums?

ecstrema, 2025-03-01:
I don't think FST support enums. 

VCD has an unofficial extension to support it, GHW supports it (it's basically binary vcd), but from I remember from the spec, it is not supported in FST. 

Gtkwave has had it that way for a long time now: you specify the enum values manually.

ArcSpecter, 2025-03-01:
ok I did a quick experiment where the Iverilog vvp command was ran with `-fst` flag and it produced fst instead of vcd waveform, then in GTKWave I opened it, added same signal `s_current` and used same textual file attached in my previous post, and GTKWave did again display text for FSM states inside its waveform UI.

ecstrema, 2025-03-02:
There currently is a technical committee forming up to create a new waveform format. The goal of it is to improve with these types of things over FST.

Lramseyer, 2025-03-02:
> There currently is a technical committee forming up to create a new waveform format. The goal of it is to improve with these types of things over FST.

Could you get me in contact with these people? I can also let the Surfer devs know as well if they're interested. I have worked with large waveform dumps in the past, and I am quite aware of what the pain points are with large FST files. FST is a great format, but it has some rather annoying limitations when dealing with large netlists.

Lramseyer, 2025-03-03:
Yes, that works!

ecstrema, 2025-03-03:
> > Sure, can I use your profile email to add you to the email thread?
>
> Yes, that works!

Great. Under what name should I introduce you?

ekiwi, 2025-03-07:
> I don't think FST supports enums.

FST does, in fact, support enums. Verilator uses that feature: https://github.com/verilator/verilator/blob/9b4509f7d90f95c135da65ec5cf1f203915897d7/include/verilated_fst_c.cpp#L123

Wellen exposes this data as: https://docs.rs/wellen/latest/wellen/struct.Var.html#method.enum_type

closed 2025-11-05 by Lramseyer