Lramseyer/vaporview #41: wit2ts --outDir ./src/extension_core ./wit :  Failed to process document TypeError: Cannot convert undefined or null to object

heyfey, 2025-02-21:
Hi, thanks for the great extension!

I tried to build the project (1.3.0) from scratch but got the below error in wit2ts:

```
$ npm run generate:model

> vaporview@1.3.0 generate:model
> wit2ts --outDir ./src/extension_core ./wit

Failed to process document
TypeError: Cannot convert undefined or null to object
    at Function.values (<anonymous>)
    at checkForResultError (/home/heyfey/git-repos/vaporview-fsdb/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:611:41)
    at new SymbolTable (/home/heyfey/git-repos/vaporview-fsdb/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:657:21)
    at DocumentEmitter.build (/home/heyfey/git-repos/vaporview-fsdb/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:1787:25)
    at processDocument (/home/heyfey/git-repos/vaporview-fsdb/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:37:21)
    at run (/home/heyfey/git-repos/vaporview-fsdb/node_modules/@vscode/wasm-component-model/lib/tools/main.js:96:50)
```

```
$ cargo --version
cargo 1.85.0 (d73d2caf9 2024-12-31)

$rustc --version
rustc 1.85.0 (4d91de4e4 2025-02-17)

$ wasm-tools --version
wasm-tools 1.226.0

$ wit2ts --version
1.0.1
```

OS: Ubuntu 22.04 (WSL2)
Commit: 0200c87f28a6a13900fdf767fd3cddae47d7f4c9

Do you have any suggestions?

Lramseyer, 2025-02-21:
This is a rather strange issue. I was actually able to recreate it, but it looks like it's an issue with the wit2ts utility. It looks like wit2ts 1.0.1 is not playing nicely with the latest version of wasm-tools. I need to do the lovely job of figuring out which version of which package broke this, and file a bug report on the respective repositories in question. Once I do that, I will update the getting-started docs to warn users not to use the latest version, and update this issue with the workaround!

Also, if you're planning on contributing, my contributing doc is a little out of date AND I am doing a rewrite of the viewport component of the webview (Using Canvas instead of SVGs.) But if you want, I can help explain any area of the code base you're interested in! Send me an email if you're interested!

Lramseyer, 2025-02-22:
OK, so here's my working environment

`lloydramseyer@Lloyds-MBP vaporview % cargo --version`
`cargo 1.82.0 (8f40fc59f 2024-08-21)`
`lloydramseyer@Lloyds-MBP vaporview % rustc --version`
`rustc 1.82.0 (f6e511eec 2024-10-15)`
`lloydramseyer@Lloyds-MBP vaporview % wasm-tools --version`
`wasm-tools 1.220.0`
`lloydramseyer@Lloyds-MBP vaporview % node_modules/@vscode/wasm-component-model/bin/wit2ts -v`
`1.0.1`

But it gets worse! I tried rolling back wasm-tools, but I am getting compile errors. I might need to roll back cargo to get it to work.

I filed an [issue](https://github.com/microsoft/vscode-wasm/issues/220) in the wit2ts repo.

Lramseyer, 2025-03-05:
OK, so the issue was fixed in the vscode-wasm repo. I'm not sure which build it will make it into, so we're probably going to have to wait a little longer to mark this as complete.

closed 2025-06-14 by Lramseyer