Lramseyer/vaporview #44: Copy Selection as Wavedrom not working

epsilon537, 2025-02-27:
**Describe the bug**
On my system, the 'Copy selection as Wavedrom' feature doesn't leave a Wavedrom JSON on the clipboard.

**To Reproduce**
1. Open a .vcd
2. Select three signals.
3. Set marker and alt-marker across a span of about 8 clock cycles.
4. Select clock signal and select 'Set WaveDrom clock Rising'.
5. Select 'Copy selection as WaveDrom'.
6. In a new text file in the same VS Code session, select 'Paste'.

**Expected behavior**
I expected to see the Wavedrom JSON data in the text file.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04.2 LTS
 - Version: VS Code 1.97.2, VaporView 1.3.0.

Lramseyer, 2025-02-28:
OK, give the [1.3.1 Pre-Release](https://github.com/Lramseyer/vaporview/releases/tag/1.3.1-pre-release) a try to see if it fixes the issue.

epsilon537, 2025-02-28:
Thanks for the quick response. I'm having some issues getting through the build procedure. This step is failing:

```
> vaporview@1.3.0 generate:model
> wit2ts --outDir ./src/extension_core ./wit

Failed to process document
TypeError: Cannot convert undefined or null to object
    at Function.values (<anonymous>)
    at checkForResultError (/home/epsilon/work/vaporview-1.3.1-pre-release/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:611:41)
    at new SymbolTable (/home/epsilon/work/vaporview-1.3.1-pre-release/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:657:21)
    at DocumentEmitter.build (/home/epsilon/work/vaporview-1.3.1-pre-release/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:1787:25)
    at processDocument (/home/epsilon/work/vaporview-1.3.1-pre-release/node_modules/@vscode/wasm-component-model/lib/tools/wit2ts.js:37:21)
    at run (/home/epsilon/work/vaporview-1.3.1-pre-release/node_modules/@vscode/wasm-component-model/lib/tools/main.js:96:50)
```

Lramseyer, 2025-02-28:
Yeah, there's a [bug](https://github.com/microsoft/vscode-wasm/issues/220) in wit2ts. Just download the .vsix file. I forgot to upload it last night, but I just uploaded it in the [releases](https://github.com/Lramseyer/vaporview/releases) section.

closed 2025-02-28 by epsilon537