Lramseyer/vaporview #48: Support for multidimensional arrays

Lawrence-lugs, 2025-03-07:
Vaporview:
[image]

Hello! Thanks for this great extension.
unpacked arrays don't show up in the view. Is this an unimplemented feature?
Structs seem to show up as intended.

I'm using Vaporview in conjunction with remote-SSH.
(Also notice my arrays do not have bitwidth indicators like [31:0], which might be a related problem)

Lramseyer, 2025-03-07:
What version of Vaporview are you using? That looks like it's pre 1.2.0. After 1.2.0, I use a different VCD parser, which might fix your problem. If not, let me know, and we can debug from there!

Lawrence-lugs, 2025-03-07:
Thanks for pointing that out! It turns out my vscode was outdated due to a MacOS bug so the extensions weren't up to date.
Now, I'm using 1.3.1, and it seems to be supported.

closed 2025-03-07 by Lawrence-lugs