Lramseyer/vaporview #52: Colors are not being loaded properly from save file

Diogo-Valadares, 2025-03-12:
**Describe the bug**
After executing a "Save Vaporwave Settings" and then "Load Vaporwave Settings" the colors are sometimes loaded wrong, E.G. an orange signal goes back to the default color.

**To Reproduce**
Steps to reproduce the behavior:
1. Add signals
2. Set different colors
3. Save Vaporwave Settings
4. Load Vaporwave Settings

**Expected behavior**
The colors should not change after loading

**Screenshots**
Its really not consistent but one example

I saved like this:
[image]

The A bus wire color changes to cyan
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - vscode 1.98.1
 - Version 1.3.1

Lramseyer, 2025-03-13:
Do you know if this is a problem when you save the file, or when you load the file? Do you have the settings file you used for this example?

Diogo-Valadares, 2025-03-13:
After saving, looking at the .json of the example below, the color index of the c_address is wrong so I'm guessing the problem is when I save the file.

```
{
  "netlistId": 160,
  "name": "test_drisc.drisc_processor.register_file_0.c_address",
  "numberFormat": "hexadecimal",
  "colorIndex": 0,
  "renderType": "multiBit"
},
{
  "netlistId": 59,
  "name": "test_drisc.drisc_processor.c_bus",
  "numberFormat": "hexadecimal",
  "colorIndex": 2,
  "renderType": "multiBit"
},
```

|Saved like this:|Loaded like this:|
|----------------|-----------------|
|[image]|[image]|

save file:
[test.json](https://github.com/user-attachments/files/19229946/test.json)
vcd:
[dump.txt](https://github.com/user-attachments/files/19229953/dump.txt)

Lramseyer, 2025-03-14:
OK first of all, excellent find and excellent debug work! This helps a ton!

I need to take a look at this tonight, but when loading in a save file it looks up the variables by path name _specifically_ so that it won't break if you change the circuit. So if it's breaking when you change your design (assuming path names of what's in the viewer don't change) then that defeats the purpose, and I need to fix that!

Lramseyer, 2025-03-14:
OK, I think I fixed this in the latest commit. I was able to recreate the save file not saving off the color correctly, which I was able to fix. However, you also mentioned another issue, where when the circuit changes, things don't load correctly. I found what I believe to be the root cause of it and fixed it, but unfortunately, I don't have any test files to test that out on. So I'm going to need some help. I updated the [release](https://github.com/Lramseyer/vaporview/releases/tag/1.3.2-release-candidate) build to include these changes in case you want to try it out!

Diogo-Valadares, 2025-03-14:
Well, that was quick! 
The saving seems to be fixed, but the loading is still broken

The files below should help you testing
Config file:
[test.json](https://github.com/user-attachments/files/19253131/test.json)

vcd that doesn't load correctly
[vcd.txt](https://github.com/user-attachments/files/19253138/vcd.txt)

vcd that loads correctly
[working_vcd.txt](https://github.com/user-attachments/files/19253190/working_vcd.txt)

Lramseyer, 2025-03-14:
>but the loading is still broken

Are you referring to the issue where a different circuit has an incompatible save file? Or is the original issue of colors not loading correctly still an issue?

Also, I assume that the test.json was created from working_vcd.txt?

Diogo-Valadares, 2025-03-14:
> Are you referring to the issue where a different circuit has an incompatible save file

Yes.

> Also, I assume that the test.json was created from working_vcd.txt?

Yes, the working_vcd was the one used to create the save file, and the vcd.txt is a modified version of this circuit. The signals being tracked are the same, so the colors and other info were supposed to load correctly but they aren't.

Lramseyer, 2025-03-17:
Thanks for the test files! I was able to recreate the issue and fix it. I updated the [release](https://github.com/Lramseyer/vaporview/releases/tag/1.3.2-release-candidate) build again with this fix.

Diogo-Valadares, 2025-03-17:
It does seem to be working properly now. At first it wasn't working and spent some minutes trying to find out what was wrong but I just had to restart VScode and now it works like a charm.

I'll keep testing and let you know if there is anything broken. 

Thanks for creating this wonderful extension!

Lramseyer, 2025-03-19:
1.3.2 has been released. I'm going to close this issue if nobody is experiencing this issue anymore, but I will give it a few more days.

closed 2025-03-26 by Lramseyer

victor-fisyuk, 2025-09-13:
Hi everyone,
I'm experiencing a similar issue where `numberFormat` and `renderType` break after loading saved settings whenever I add new signals to the modules (even in the latest version `1.4.1`).
Thank you.

[image]

[image]

[image]