Lramseyer/vaporview #53: Ability to adjust the timestamps

Nebu0528, 2025-03-13:
Would it be possible to change how the timestamps are being displayed, perhaps in different units or formats? I notice sometimes it gets displayed in decimals when I toggle the viewer.

If so, please let me know and would love to contribute :)

Lramseyer, 2025-03-13:
The ruler was one of the first things I created when designing Vaporview. It may be due for a redesign. I'm open to ideas, but it should _feel_ like something that lives inside of VScode, or a waveform viewer in the design style of [Monaco](https://microsoft.github.io/monaco-editor/).

Nebu0528, 2025-03-14:
Sounds good, I'd be happy in contributing to the feature if you are open to it.

Lramseyer, 2025-03-14:
I am interested in more contributors! Though you will have to forgive me, as I am still learning the ins and outs of managing an open source project _with other contributors_. My email is in my profile, so feel free to email me directly if you need. There's also a discussions section, which might be useful. Either one works!

Right now, I am in the process of finalizing the 1.3.2 release so I can get it out to the marketplace.

Nebu0528, 2025-03-14:
Perfect, looking forward to the release.

Lramseyer, 2025-04-20:
OK, I just added this feature and checked it in. It will be available in the next release!

Lramseyer, 2025-05-09:
OK, I think I implemented what you asked for in version 1.3.4. It's now generally available on the [VScode Marketplace](https://marketplace.visualstudio.com/items?itemName=lramseyer.vaporview). But if there's something else you want to see, feel free to open up another issue or discussion!