Lramseyer/vaporview #56: Add separators/blank spaces

elpollodiablo84, 2025-03-19:
A feature that I find very useful in other viewers is the possibility to add some separator/blank space between signals to help navigate visually a waveform with a lot of items. Grouping signals under a common node/folder is also a nice feature to have, but I think it's more difficult to implement and not particularly needed right now.
Thanks for this nice project!

Lramseyer, 2025-03-19:
This was actually in the plan for 1.4.0, so this validates my plan! I might do a 1.3.3 release depending on how easy some of the other features are to implement, but adding in the feature to add a separator/blank space should be fairly easy in the short term.

> I think it's more difficult to implement and not particularly needed right now.

It was difficult to implement a waveform viewer extension for VScode, and nobody _needed_ it, but I still did it 😉

Lramseyer, 2025-12-12:
Marketplace release is out

closed 2025-12-12 by Lramseyer