Lramseyer/vaporview #57: Add some minor features

elpollodiablo84, 2025-03-20:
Following my last issue, here some other really small ideas that can be added whenever it feels right:

1. Add a button to fit all the wave into the screen
2. Add the possibility to change the time unit measure in the UI (right now only _fs_ if I'm not mistaken)
3. "Sticky scroll" on netlist tree view
4. Rename selected signals 

Since these are small features, I created a single issue to track them, but if needed I can create separate issues for every one of (or some of) them.

heyfey, 2025-03-20:
Here are some of my thoughts:
 
1.
Add a button to fit all the wave into the screen -> I used to want this feature, just like [100%] bottom in Verdi. But later I realize I don't need it if the zooming-out is extremely smooth.

4.
Rename selected signals -> what's the use case?

5.
Copy value of signal at cursor -> maybe add "copy variable value" just like Surfer has?

Lramseyer, 2025-03-20:
>    1. Add a button to fit all the wave into the screen

I should add this. The biggest challenge is that I need to actually make the icon to do this, and I want it to follow the design style as the [vscode codicons](https://github.com/microsoft/vscode-codicons)

>    2. Add the possibility to change the time unit measure in the UI (right now only fs if I'm not mistaken)

The time units depend on your waveform dump time units. But this is an interesting idea.

>    3. "Sticky scroll" on netlist tree view

Not sure what you mean by this. Could you elaborate Though I should point out that I am using the built in [VScode Tree View](https://code.visualstudio.com/api/extension-guides/tree-view) for my netlist viewer. When I think of sticky scrolling, I think of [sticky position](https://developer.mozilla.org/en-US/docs/Web/CSS/position), which newer versions of VScode have in their tree views.

>    4. Rename selected signals

Not sure what you're referring to here either.

>    5. Copy value of signal at cursor

Yeah, I noticed last night that I should probably add this. I think what I am going to do, is copy the value in a slightly different format than what is displayed, so as an example if a value is [hex formatted] 7000_3001, I would remove the "_" spacers when copying to clipboard. Do you think that would be a good idea?

elpollodiablo84, 2025-03-20:
> > 1. Add a button to fit all the wave into the screen
> 
> I should add this. The biggest challenge is that I need to actually make the icon to do this, and I want it to follow the design style as the [vscode codicons](https://github.com/microsoft/vscode-codicons)

Well, I can _really_ understand the struggle for consistency 😄 The zoom in/out is really smooth, even with a lot of data, so this could be considered a secondary request.

> > 2. Add the possibility to change the time unit measure in the UI (right now only fs if I'm not mistaken)
> 
> The time units depend on your waveform dump time units. But this is an interesting idea.

Yeah I know that depends on the waveform. If I'm not mistaken ghdl saves data in femtoseconds as default (in the ghw format), so this feature could be useful.

> Not sure what you mean by this. Could you elaborate Though I should point out that I am using the built in [VScode Tree View](https://code.visualstudio.com/api/extension-guides/tree-view) for my [...]

I mean [this thing](https://learn.microsoft.com/en-us/visualstudio/ide/editor-sticky-scroll?view=vs-2022). It could be useful when you have a lot of signal inside an entity and you lose track of the parent of those signals.

> > 4. Rename selected signals
> 
> Not sure what you're referring to here either.

I mean to use a custom name instead of the signal's name in the wave viewer. I use it a lot in the Vivado simulator when I need to 
check some usually obscure code written by other people and I need to rename signals in the viewer to understand better what is happening. It works right-clicking on the signal, selecting smothing like "Use custom name" in the context menu and then typing the string that will be used instead of the default name "path/to/signal/name_of_the_signal".  Howewer, IMO this feature is really of low priority, it is just a random idea from daily experience with other wave viewers.

> Yeah, I noticed last night that I should probably add this. I think what I am going to do, is copy the value in a slightly different format than what is displayed, so as an example if a value is [...]

I think the formatting doesn't really matter, there's always some way to convert the value. If it's simpler a straightforward int is more than enough!

Lramseyer, 2025-03-27:
OK, I implemented "Copy Value at Marker" and got that checked in.

Regarding sticky scrolling, it looks like this is an feature you can enable in VScode. [Here's](https://stackoverflow.com/questions/77581228/how-can-i-configure-sticky-scrolling-of-tree-views-in-vs-code) a stack overflow discussion about it.

elpollodiablo84, 2025-03-27:
I just discovered that after a recent (?) update of vscode, sticky scrolling was disabled in my installation everywhere, so  you can check also the third point, sorry 😅

Lramseyer, 2025-04-20:
The following features are implemented in the 1.3.3 release (which is available on the VScode Marketplace):
- (1) Add a button to fit all the wave into the screen
- (5) Copy value of signal at cursor
- ~(3) "Sticky scroll" on netlist tree view~ (Can be enabled in VScode settings)

Complete and checked in. Will be included in the next release:
- (2) Add the possibility to change the time unit measure in the UI (right now only fs if I'm not mistaken)

Rejected for now:
- (4) Rename selected signals

Once the next release is available that implements the fix for item 2, I will close this issue.

Lramseyer, 2025-05-09:
Item 2 has been implemented, and is available in version 1.3.4. It's is now generally available on the [VScode Marketplace](https://marketplace.visualstudio.com/items?itemName=lramseyer.vaporview), so I am closing this.

closed 2025-05-09 by Lramseyer