Lramseyer/vaporview #61: API

qarlosalberto, 2025-03-29:
Could you add a command with argument the waveform path? I would like to call it from my extension with:

```
const commandID = "surfer.openWaveform";
await vscode.commands.executeCommand(commandID, waveformURIPath);
```

Lramseyer, 2025-03-30:
Vaporview automatically associates VCD, FST, and GHW files, so the general open file command should open up vaporview if that's easier. I can still add this command if you would like though.

Since you're here... I'm putting together an API documentation to try to standardize interoperability between extensions. I noticed that Teros HDL supports RTL elaboration, so it should be pretty easy to allow vaporview to connect back to an instance path in the RTL. This way, a signal in the waveform viewer can point back to a location in the RTL. Similarly, you should be able to (with a designated instance selected) add a signal into the waveform viewer via a variable in the RTL. Verdi does this, but Synopsys licenses are not cheap. Wouldn't it be nice if we could give that feature to everyone for free?

I'll keep you posted with the API docs. I'll also create test builds for you so that you can test these out!

qarlosalberto, 2025-04-01:
Great! Could you add "waveformViewer.open"? Because maybe the user has multiple viewers.

Lramseyer, 2025-04-09:
Just an update to others out there - there has been offline discussion about the API. If any of you have any suggestions for what I should add or modify, feel free to discuss here!

I'll be doing a 1.3.3 release once I clean up variable name handling and instance paths. It will have the preliminary API implemented. If you plan to implement any of the Vaporview features in your extension, please let me know! The 1.4 release will use a finalized version of the API.

[API docs](https://github.com/Lramseyer/vaporview/blob/main/API_DOCS.md)