Lramseyer/vaporview #64: Add automatic reload

pelrun, 2025-04-28:
Standard document behaviour in vscode is to reload automatically when the file changes on disk. Waveforms views currently do not. At the least, it should be a configurable setting.

Lramseyer, 2025-04-28:
I use a parsing library to load waveform dumps, and to reload a document, it would need to re-parse the netlist hierarchy. For small files, this won't be a problem. But for larger files, it takes a few seconds to parse the netlist.

If I'm able to do this in the near term, I will probably set a netlist size threshold that can be configured in the user settings. But I do plan on allowing the viewport to extend the time.

pelrun, 2025-04-29:
Size would work, but a simple checkbox in the UI to turn it on and off on a per-file basis would be nice too - then it'd be easy to turn it off when it's causing issues.

Lramseyer, 2025-05-01:
I haven't forgotten about this, and I will be looking into this issue when I get back from Latch Up. But I should point out (in case you're unaware) that you can right click on a tab and select "Reload File" to manually reload. I know that's not exactly what you're asking for, but hopefully it gets you by in the meantime!

Also, I think the threshold I will set will be based on time it takes to load the file rather than netlist size. I'm hesitant turn this on by default (but I'm happy to be convinced otherwise) because default behavior _is_ the behavior for about 80% of users, and I want to make sure that the default behavior makes sense to them.

ryankurte, 2025-06-01:
definitely be neat to have an auto-reload option! FYI for some reason selecting "Reload File" doesn't seem to actually reload the wave file for me.

vbidula, 2025-06-12:
Came here to second this idea. I would absolutely love it if newly generated waves could auto-reload with the same settings, colors, and cursor position.

That said, thanks a lot for this tool – it is super useful and really well made.

ryankurte, 2025-06-21:
confirming the "reload file" option also doesn't work on macOS, so, i suspect this is just broken.

another option would be to automatically load a configuration matching the name of the wave (so if `something.json` exists when `something.vcd` is loaded it would auto apply these settings), or to automatically persist wave configs using a shadow config (`.something.json`) file.

Lramseyer, 2025-08-21:
Good news! This feature will be in the 1.4.0 release. You can download test builds with the feature enabled [here](https://github.com/Lramseyer/vaporview/releases/tag/1.4.0-rc).

Lramseyer, 2025-08-28:
1.4.0 is released and on the VScode marketplace, so I will close this feature!

closed 2025-08-28 by Lramseyer