Lramseyer/vaporview #66: X signal Mis-rendering

skyhgzsh, 2025-05-08:
**Describe the bug**
Mis-rendering x signal when zoom the wave diagram in a certain size.

**To Reproduce**
Steps to reproduce the behavior:
1. open "dump.fst"
2. Load "vaporview.json"
3. Zoom to a certain size as the picture show
4. See error

**Expected behavior**
the x signal as red line should show between 730 and 750, not before 730 (ns)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser : VSCode
 - Version 1.3.3

**Additional context**

[image]

[vaporview.json](https://github.com/user-attachments/files/20097558/vaporview.json)

[dump.json](https://github.com/user-attachments/files/20097579/dump.json)
(*due to github upload restriction, dump.fst can only be upload like this.)

Lramseyer, 2025-05-08:
That is indeed a renderer glitch! Let me go fix that before putting 1.3.4 on the marketplace...

Lramseyer, 2025-05-08:
OK, I just pushed a fix and an updated build with said fix. Check out the [releases](https://github.com/Lramseyer/vaporview/releases/tag/1.3.4-rc) section if you want to download a build with the fix implemented!

Lramseyer, 2025-05-09:
This is fixed in version 1.3.4, which is now generally available on the [VScode Marketplace](https://marketplace.visualstudio.com/items?itemName=lramseyer.vaporview).

closed 2025-05-09 by Lramseyer