Lramseyer/vaporview #83: Analog view with fixed point support

eren-akbayin, 2025-08-10:
It would be cool to have analog view with fixed point shifting would be quite cool for digital signal processing related waveforms.

Lramseyer, 2025-09-05:
Good news! I was able to implement one of your feature requests - Signals can now have a variable height. You can test it out in the [beta build](https://github.com/Lramseyer/vaporview/releases/tag/v1.4.1-beta) and let me know what you think. Technically, digital signals and multi-bit signals can also be set taller as well (which looks a little obnoxious, but it's more work to limit users.)

eren-akbayin, 2025-09-05:
That is good news. I have tried it and it looks nice. Exactly how I imagined. Maybe it can automatically adjust the y axis' minimum and maximum to the what is shown on the screen. Otherwise it is a bit hard to see the details when the data type's range is much bigger than the currently shown data on the screen or am I missing something? Can the vaporview already do that?
[image]
Anyways looks quite cool and I appreciate the time you have spent on this.

Lramseyer, 2025-09-05:
Oh no! I knew that this would come up eventually... OK, so you can view analog signals from multi-bit signals or from Real types. For Real types, it sets the vertical scale by looking at the min and max value with either the top or bottom anchored to 0, but for multi bit signals, it sets the vertical scale by the bit range. Let me see what I can do about adjusting vertical scale. I'll probably make it a right click menu item and key binding. I will need a zoom in. zoom out, and reset zoom. This shouldn't be too hard to implement...

Lramseyer, 2025-09-05:
And it looks like I introduced a bug into the vertical culling in the renderer, so zooming and scrolling may be glitchy. I'll update this when it's fixed.

Lramseyer, 2025-09-07:
OK, try the new version I uploaded to the [releases](github.com/Lramseyer/vaporview/releases/tag/v1.4.1-beta) section. I need to clean it up a little by adding right click menus and documenting the changes, but you should be able to do what you need to do with Alt + and Alt - to change the vertical scaling.

I'm also going to add a reset vertical scale, make it part of the save file, and make sure that the vertical scale options only appear when the user has a signal displayed as an analog signal. That will go into the final version of 1.4.1, and I should have that done later this week.

eren-akbayin, 2025-09-07:
Yes, it works! One small comment though, for my keyboard it does not work with numpad's "+" and "-" but it does work with the other "+" and "-",. Can you please add also numpad ones?

Lramseyer, 2025-09-13:
1.4.1 release is out on the VScode marketplace! Can you make a separate issue for fixed point support?

eren-akbayin, 2025-09-13:
Yes, of course. Thanks for implementing analog waveforms.

closed 2025-09-13 by eren-akbayin