Lramseyer/vaporview #86: Zoom fit hotkey and multi-signal selection

hzs268, 2025-08-13:
Add a keyboard shortcut (e.g., "F") to quickly zoom the whole waveforms to fit the visible area.

Allow users to hold Ctrl (or Cmd on macOS) to select multiple signals in the signal panel, then:
     Drag-and-drop to regroup.
     Delete/disable selected signals in bulk.
     Change the signals color...

Lramseyer, 2025-08-13:
Multi signal select is definitely on the to-do list.

The zoom to fit hotkey is a great idea, and super easy to add. I'll include that in the next release, but I don't think "F" is the best shortcut key. Probably Ctrl + 0 (Command + 0 on Mac) since that's the hotkey to reset zoom level. Thoughts?

Lramseyer, 2025-08-14:
OK, I added the Ctrl/Cmd + 0 command last night. It will be in the next release.