Lramseyer/vaporview #89: Support Undo/Redo

Lramseyer, 2025-08-22:
Allow users to undo and redo actions like marker placement, signal selection, signal reordering, and adding/removing variables.

rknastenka, 2026-01-02:
Valid point about integrating `CustomEditorProvider` vsc API. I was thinking about a simpler approach though: using the Command Pattern....why?

1. It'll work immediately without refactoring `CustomReadonlyEditorProvider` or touching the file loading architecture.
2. Adding new undoable actions is straightforward without implementing `save()`, `revert()`, `backup()` semantics
3. Avoiding document model complexity.

i understand the trade offs, but for viewer state management, I think this gives us 80% of the value with 20% of the complexity. Users get Ctrl+Z/ (or wtv commands you choose) working for all major actions without the overhead of treating UI state as document edits.

Lramseyer, 2026-01-02:
A lot of the underlying functionality for save() revert() and backup() is already implemented in other functions. The big challenge with this is efficiently applying the viewer state. Whether it's implemented via CustomEditor, or by custom commands (which may be the only possible way to do it) I need to ensure that the state can be reapplied without memory leaks, and without having to recompute number formats.

If it's implemented via commands, I would also have to create my own undo/redo buffer. Not that big of a deal, but still extra work.

My goal here is ultimately to provide a user experience that is consistent with VScode, and then for code cleanliness and maintainability.